adyanth/openwrt-tailscale-enabler #1: /tmp/tailscale: line 1: syntax error: unexpected "("

tediorelee, 2021-04-10:
hello:
when I'm using tailscale on my openwrt router, I followed your docs and ran into this error
is there any way to solve it ?

[image]

thanks

adyanth, 2021-04-10:
This repo was targeted towards mips based processors running openwrt. What are you running it on? 

The output of `uname -m` would give you that. Use your processor architecture in the version field of [/usr/bin/tailscale](https://github.com/adyanth/openwrt-tailscale-enabler/blob/2082a5c0e8e14540c9224351182eb2041b1613a9/usr/bin/tailscale#L3) and [/usr/bin/tailscaled](https://github.com/adyanth/openwrt-tailscale-enabler/blob/2082a5c0e8e14540c9224351182eb2041b1613a9/usr/bin/tailscaled#L3) files.

The valid architectures for tailscale are: 386, amd64, arm, arm64, mips, mipsle. See the [tailscale package repo](https://pkgs.tailscale.com/stable/#static) for info.

tediorelee, 2021-04-12:
thanks for your reply, after I change the version field based on my processor architecture, `version='1.6.0_amd64'`, and I repacked the .tgz file, after all of this I get this error
[image]

adyanth, 2021-04-12:
There is no need to repackage it as a tar. 
It seems to be that the tailscaled service has not started, so the tailscale cli cannot connect to its socket. Sadly, the init system wont show much when it fails, but you can check the dmesg to see if anything went wrong starting the tailscale daemon.
Can you try to use the same tar but then edit the files in openwrt and then follow the rest of the steps? 

Also, you seem to have enough space in disk running on an x64 platform. This was for running it on routers whose flash is not large enough for the tailscale binary. You can actually extract the binaries from here https://pkgs.tailscale.com/stable/#static and copy them to /usr/bin/tailscale and /usr/bin/tailscaled instead of using the bootstrap scripts I wrote. Then, you can proceed to use the init scripts from here since procd scripts are not provided by tailscale.

tediorelee, 2021-04-14:
> There is no need to repackage it as a tar.
> It seems to be that the tailscaled service has not started, so the tailscale cli cannot connect to its socket. Sadly, the init system wont show much when [...]

thanks a lot, I will try this evening and to see if it works😊

closed 2021-04-23 by tediorelee