adyanth/openwrt-tailscale-enabler #4: Fails if no internet on boot

adyanth, 2021-05-29:
If the router boots without internet, the tailscaled never starts. I am not sure if there is a possible workaround other than waiting for a ping to succeed before starting the download.

marlonfan, 2022-05-08:
I added a new file to enable it and it worked!

```bash
#!/bin/sh /etc/rc.common

# Added at /etc/init.d/tailscale_up

USE_PROCD=1
START=100
STOP=1

start_service() {
  procd_open_instance
  procd_set_param command /usr/bin/tailscale

  # Set the port to listen on for incoming VPN packets.
  # Remote nodes will automatically be informed about the new port number,
  # but you might want to configure this in order to set external firewall
  # settings.
  procd_append_param command up
  procd_append_param command --advertise-exit-node
  procd_append_param command --advertise-routes=192.168.100.0/24
  procd_append_param command --accept-dns=true

  procd_set_param respawn
  procd_set_param stdout 1
  procd_set_param stderr 1

  procd_close_instance
}

stop_service() {
  /usr/bin/tailscale down
}
```

lanrat, 2022-06-23:
I fixed this in [my fork](https://github.com/lanrat/openwrt-tailscale-enabler). However I did a pretty large rewrite.

The binaries are saved to a persistent folder (or /tmp) and are only updated when running `tailscale-install`.

adyanth, 2022-06-24:
Hey @lanrat that will only work if you can save it to a persistent directory. The main problem solved by this repo is that there is no space in flash to do that.

If you have space, consider using the opkg package for tailscale which is smaller in size too I believe.

closed 2022-10-06 by adyanth