adyanth/openwrt-tailscale-enabler #5: Unable to download binares from tailscale

philippdzm, 2021-07-26:
Problem: 
Trying to download https://pkgs.tailscale.com/stable/tailscale_1.8.5_armv7l.tgz returns 404 by tailscale's server.

```
~# tailscale up --accept-dns=false --advertise-routes=10.0.0.0/24
Downloading Tailscale 1.8.5_armv7l ..
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.8.5_armv7l.tgz'
Connecting to 167.172.11.40:443
HTTP error 404
tar: invalid magic
tar: short read
mv: can't rename '/tmp/tailscale_1.8.5_armv7l/*': No such file or directory
Done!
/usr/bin/tailscale: line 29: /tmp/tailscale: not found
```

This will break everytime tailscale builds new versions as they only provide the newest at their server.

adyanth, 2021-07-26:
Yes indeed. I was assuming that it will exist, but apparently not. Only the latest stable and unstable builds.

People at Tailscale hope to have historical versions available on pkgs.tailscale.com soon: https://tailscale.com/kb/1032/install-files-all-versions/

Meanwhile, I can look into making the script check for the latest version and use that.

keyakko, 2021-08-14:
I had the same problem. But, I avoided the problem by adding the temporary change to the script `/usr/bin/tailscale`.

```
# diff /usr/bin/tailscale /tmp/usr/bin/tailscale
9c9
< version="${tailscale_version}_arm${endianness}"
---
> version="${tailscale_version}_${arch}${endianness}"
```
(`/tmp/usr/bin/tailscale` is the original source code(v1.8.5-0897f11).)

In my environment, just worked.
[image]

I think the script needs to add an architecture name check.
In my environment, the result of `uname -m` is like below.
```
# uname -m
armv7l
```
`armv7l` is [32bit processor](https://askubuntu.com/questions/928227/is-armv7l-32-or-64-bit). It is expected that `tailscale_1.8.5_arm.tgz` will be downloaded, but it will attempt to download `tailscale_1.8.5_armv7l.tgz`.

keyakko, 2021-08-14:
I added a change to `/usr/bin/tailscaled` that had a similar effect. I just had a successful authentication.
```
# diff /usr/bin/tailscaled /tmp/usr/bin/tailscaled
3c3
< arch=arm
---
> arch=`uname -m`
```
(This change is cleaner than the change I made in the previous post... I'm sorry for the disorganized message.)

adyanth, 2021-08-14:
Your issue is due to running it on an ARM processor where the `uname -m` does not match the package name. Hardcoding the architecture would not help much. Could you send a PR to specifically check for armv7l and set the arch to arm if so?

keyakko, 2021-08-14:
I just created PR now. Please checkout it.
#7

adyanth, 2021-08-14:
For the initial problem in this issue, something like this works:
```
curl -q https://pkgs.tailscale.com/stable/ | grep tailscale_ | head -1 | cut -d'_' -f 2
```

Thinking of whether to implement this in the shell script (and breaking the auto releaser github workflow in the process) or add this as a workflow cron to update every other week or so.

PR #8 (2021-08-15): ✨ Auto updating shell script and Github release
Fixes #5
closed 2021-08-15 by adyanth