adyanth/openwrt-tailscale-enabler #6: Lost in update process

mrogojanu, 2021-08-07:
Hi,

I've managed to install tailscale on Archer C7 v5 (128MB ram / 16MB flash).  So far so good and looking forward to update it as being notified in the admin panel for using an older version.  The current version on https://pkgs.tailscale.com/stable/#static  seems to be tailscale_1.12.3_mips.tgz  while the one installed on router is:  

root@OpenWrt:/usr/bin# tailscale --version
1.8.5
  tailscale commit: 118cf0cd75f92bbe3d994a5b1587c9ea3553221b
  other commit: c6fd9c9d5657e82c2d7888804ebce0e8bd6a4556
  go version: go1.16.4-ts1b677aa

, on git hub page the reference is for version="1.2.10_mips" which seams be higher than 1.12.3 as last version on download page. Confused here. 

Moreover the size of tailscale/tailscaled in tailscale_1.12.3_mips.tgz package is around 10MB each file while the one installed with openwrt-tailscale-enabler is around 1KB each.   So I am probably looking at wrong packages the openwrt update.  Kindly request some guidelines.

adyanth, 2021-08-07:
Tailscale does not provide old version downloads (yet). This is a pain, and the only workaround for now is to set the version to the one available on pkgs.tailscale.com.

Also, 1.2.10 is not higher than 1.12.3. 1.20 would be. You can set the version to 1.12.3 in both `/usr/bin/tailscale` and `/usr/bin/tailscaled`, and it should run just fine. Refer https://github.com/adyanth/openwrt-tailscale-enabler/issues/5

Regarding the size, you are probably looking at the file `/usr/bin/tailscale[d]` which is just a bash file. This will download the actual 10+ MB binary to `/tmp/tailscale[d]` and run it from there, since the `/tmp` lives on RAM where we have space.

mrogojanu, 2021-08-07:
Thank's  for quick feedback.  Manual editing the versions resolved the issue.

closed 2021-08-07 by mrogojanu