adyanth/openwrt-tailscale-enabler #15: Doesn't work for devices with 64MB of RAM

bam80, 2021-11-07:
I tried different things (zram, etc.) but still couldn't make it work with 64MB of RAM - even if the device manages to download the tailscale package to get the binaries to /tmp, it become practically unusable due lack of memory and `tailscale up` can't finish.

Do both of `tailscale` and `tailscaled` binaries are necessary? Can we skip one of them during the installation?

adyanth, 2021-11-07:
Hey @bam80 I have a device with 128MB RAM and it definitely uses more than 64MB.

Once the Tailscale up is run the first time, you can delete the Tailscale binary, leaving just the daemon running. It is only needed for configuration.

bam80, 2021-11-07:
Hey! Unfortunately, it can't even pass the `tailscale up` phase as the device gets stalled.. :(
So I don't know workaround for this, other than adding more memory by any mean or using remote mount.
If you have better idea please let me know.

adyanth, 2021-11-07:
Umm, maybe try copying over the `tailscale.state` file from another device after performing `tailscale up` ? I am not sure if it would work because I remember it uses some device unique ID.

bam80, 2021-11-07:
Thanks, this is a bright idea!
Until we didn't test it/have not instructions - should we edit Readme a bit to note about the RAM restrictions?

JensKuehnel, 2021-11-07:
Hi,
I have tailscale running on two 64MB boxes. And it runs without any problems.
I build my own images with image-builder and running 21.2.1 on a "Netgear WNDR3700" and a "TP-Link CPE210 v3"

This is my PACKAGES list: 
```
diffutils luci-ssl zram-swap diffutils luci-ssl curl vim-full bash -vimi mtr ethtool tcpdump -luci-proto-ppp -ppp -ppp-mod-pppoe block-mount kmod-fs-ext4 kmod-usb-storage kmod-usb-ohci kmod-usb-uhci ca-bundle kmod-tun
```

No special configuration, it simply works.

@bam80 Have build the you image your self, any extra packages installed? What Version of OpenWRT?

bam80, 2021-11-07:
Thanks for reply @JensKuehnel!
I run vendor firmware based on OpenWRT 19.07, the Tailscale package was not backported there so I created a request:
https://github.com/openwrt/packages/issues/17088
Meanwhile, do you think I could easily build the package myself? I'm not sure where to start from..

(Also, you probably don't need to build custom image as the package is already there for your OpenWrt version: https://github.com/openwrt/packages/pull/15539)

adyanth, 2021-11-08:
Thats a great idea, you could try compiling it yourself and see if it fits then. Also, the package on openwrt installs tailscale to flash rather than RAM, so if you have enough flash (>32MB) and less RAM, definitely go that route.

bam80, 2021-11-09:
Well, I tried on another router with 128MB RAM/8MB flash so no chances to fit packaged Tailscale there also:
[image]
Why it so big I wonder? Any chance to rebuild it smaller?

Also, even with 128MB RAM it couldn't run, so I had to install `zram-swap` package to solve that.
I think it worth to be mentioned in Readme.

bam80, 2021-11-16:
> Also, the package on openwrt installs tailscale to flash rather than RAM

I wish I could force it to install to RAM, as it doesn't fit to flash but packaged version is still smaller..

adyanth, 2022-04-10:
The difference in size between the ipk version and the one this package downloads is shown below. I downloaded the latest version available on both.

IPK Version: 14M after extraction (even though the ipk size is 4.9MB), dynamically linked binary, depends on libc.
```bash
❯ file tailscaled
tailscaled: ELF 32-bit MSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-mips-sf.so.1, Go BuildID=1649416067, stripped
❯ ls -alh tailscaled
-rwxr-xr-x@ 1 adyah  staff    14M Dec 20 14:57 tailscaled
```

This package version: 20MB, statically linked binary with no dependencies.
```bash
❯ file tailscaled
tailscaled: ELF 32-bit MSB executable, MIPS, MIPS32 version 1 (SYSV), statically linked, Go BuildID=mQBLllo7XjQZM7G-WI-4/AYM9SvzjrByCuadLpzlE/79z0Ilt18IDEc6vdrteF/UBXBXKqgGyLlaD1Y1XtY, not stripped
❯ ls -alh tailscaled
-rwxr-xr-x  1 adyah  staff    20M Apr 10 11:37 tailscaled
```

Overall, it is the same size requirements at the end of the day. Installing an ipk to RAM is not something I am familiar with. But there is nothing this package can do to reduce the size IMO. Let me know if you figure something out.

closed 2022-04-10 by adyanth