adyanth/openwrt-tailscale-enabler #20: Something weird happens with binaries

pentago, 2022-10-09:
When I extract the archive and try running any binary, I get errors like:
```
root@OpenWrt:/tmp/ts/tailscale_1.30.2_mips# tailscale -h
/usr/sbin/tailscale: line 1: syntax error: unexpected "("
```

I downloaded the proper archive I think:
`https://pkgs.tailscale.com/stable/tailscale_1.30.2_mips.tgz`

ANy idea what could be the issue?

adyanth, 2022-10-10:
That usually happens when you try to run MIPS LE binaries on MIPS or vice versa. Please check your platform. The script in this repo determines the endianness and correctly downloads the needed binaries.

pentago, 2022-10-10:
You're right, I used MIPS  instead of MIPSLE binaries. All good now, thanks!

closed 2022-10-10 by pentago