adyanth/openwrt-tailscale-enabler #22: Installation hangs while writing to stdout

SGprooo, 2022-11-03:
After running 'Tailscale up --accept-dns=false --advertise-routes=10.0.0.0/24', the download operation hangs after writing to stdout;
 ```
Downloading 'https://pkgs.tailscale.com/stable/'

Connecting to 167.172.11.40:443
Writing to stdout

Download completed (24496 bytes)
Downloading Tailscale 1.32.2_mipsle ..
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz'
Connecting to 167.172.11.40:443
Writing to stdout
tailscale_1.32.2_mipsle/tailscale
-                     24% |*******                        |  4640k  0:00:00 ETAKilled
tar: unexpected end of file
tar: short read
```
This behavior appear to be documented on [this blog](https://blog.patshead.com/2020/10/tailscale-on-my-gl-dot-inet-mango-openwrt-router.html).
The router is running a fresh installation of OpenWrt on the most recent firmware, other info below:

```
Model | Asus RT-AC51U
Architecture | MediaTek MT7620A ver:2 eco:6
Target Platform | ramips/mt7620
Firmware Version | OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd
Kernel Version | 5.10.146
```

adyanth, 2022-11-03:
That does not look to be the issue in the blog. For some reason, tar is not able to extract it.

Can you download "https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz" on your machine and use tar to extract it successfully? If so, can you run the bowl commands manually on your router and share the output? How much RAM (free) does your router have?

```bash
du -sh /tmp
df -h
mount
free -mh
wget -O- https://pkgs.tailscale.com/stable/ | grep tailscale_ | head -1 | cut -d'_' -f 2
echo -e "tailscale_1.32.2_mipsle/tailscale" > /tmp/tailscale_1.32.2_mipsle_files.txt
wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - -C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt
```

SGprooo, 2022-11-03:
> That does not look to be the issue in the blog. For some reason, tar is not able to extract the downloaded file.
> 
> Can you download "https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz" on your machine and use tar to extract it successfully? If so, can you run the bowl commands manually on your router and share the output? How much RAM (free) does your router have?
> 
> ```shell
> du -sh /tmp
> df -h
> mount
> free -mh
> wget -O- https://pkgs.tailscale.com/stable/ | grep tailscale_ | head -1 | cut -d'_' -f 2
> echo -e "tailscale_1.32.2_mipsle/tailscale" > /tmp/tailscale_1.32.2_mipsle_files.txt
> wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - -C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt
> ```

It seems like tar reads the output directly from the stdout, sees the binary "tailscale_1.32.2_mipsle/tailscale" and begins extraction while "tailscale_1.32.2_mipsle/tailscaled" is still being outputted, and hangs from that point on.
`root@OpenWrt:/tmp# wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - 
-C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt    
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz'
Connecting to 167.172.11.40:443
Writing to stdout
tailscale_1.32.2_mipsle/tailscale
-                     35% |***********                    |  6736k  0:00:12 ETAtailscale_1.32.2_mipsle/tailscaled
^\Quit
Quit
root@OpenWrt:/tmp# 
root@OpenWrt:/tmp# wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - 
-C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt    
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz'
Connecting to 167.172.11.40:443
Writing to stdout
tailscale_1.32.2_mipsle/tailscale
-                     36% |***********                    |  6912k  0:00:10 ETAtailscale_1.32.2_mipsle/tailscaled
^\Quit
Quit
root@OpenWrt:/tmp# 
root@OpenWrt:/tmp# wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - 
-C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt    
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz'
Connecting to 167.172.11.40:443
Writing to stdout
tailscale_1.32.2_mipsle/tailscale
-                     35% |**********                     |  6704k  0:00:01 ETAtailscale_1.32.2_mipsle/tailscaled
^\Quit
Quit
`
I'll test what you mentioned above and get back asap.

SGprooo, 2022-11-03:
> That does not look to be the issue in the blog. For some reason, tar is not able to extract the downloaded file.
>
> Can you download "https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz" on [...]

It seems like tar reads the output directly from the stdout, sees the binary "tailscale_1.32.2_mipsle/tailscale" and begins extraction while "tailscale_1.32.2_mipsle/tailscaled" is still being outputted, and hangs from that point on.

`root@OpenWrt:/tmp# wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - 
-C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt    
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz'
Connecting to 167.172.11.40:443
Writing to stdout
tailscale_1.32.2_mipsle/tailscale
-                     35% |**********                     |  6704k  0:00:01 ETAtailscale_1.32.2_mipsle/tailscaled
^\Quit
Quit
`

I'll test what you mentioned above and will get back asap.

adyanth, 2022-11-03:
That is correct behavior. It can start extracting files when the rest is still being piped, it will cause wget to pause for a while. But tar should not say unexpected end of file.

SGprooo, 2022-11-03:
> That does not look to be the issue in the blog. For some reason, tar is not able to extract the downloaded file.
>
> Can you download "https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz" on [...]

It seems when running ` wget -O- https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz | tar x -zvf - -C /tmp -T /tmp/tailscale_1.32.2_mipsle_files.txt `, tar reads the output directly from the stdout, sees the binary "tailscale_1.32.2_mipsle/tailscale" and begins extraction while "tailscale_1.32.2_mipsle/tailscaled" is still being outputted, and hangs from that point on.

I'll test what you mentioned above and will get back asap.

SGprooo, 2022-11-03:
> That is correct behavior. It can start extracting files when the rest is still being piped, it will cause wget to pause for a while. But tar should not say unexpected end of file.

Running the command above several times does complete without problems, but when running the single line 'wget' command, it does still hang.
I'll include the commands above into the script, I appreciate your help.

SGprooo, 2022-11-03:
It seems like the problem still occurs when running as a shell script.

adyanth, 2024-05-04:
Is this occasional or happens everytime? I havent had this problem in quite a while.

closed 2024-05-04 by adyanth