adyanth/openwrt-tailscale-enabler #23: failed to connect to local tailscaled; it doesn't appear to be running (sudo systemctl start tailscaled ?)

dianariyanto, 2022-11-06:
```
root@GDR-SVR:~# tailscale up --advertise-routes=192.168.1.0/24
Downloading 'https://pkgs.tailscale.com/stable/'
Connecting to 167.172.11.40:443
Writing to stdout

Download completed (24496 bytes)
Downloading Tailscale 1.32.2_mipsle ..
Downloading 'https://pkgs.tailscale.com/stable/tailscale_1.32.2_mipsle.tgz'
Connecting to 167.172.11.40:443
Writing to stdout
tailscale_1.32.2_mipsle/tailscale
-                    100% |*******************************| 18924k  0:00:00 ETA
Download completed (19378646 bytes)
Done!
failed to connect to local tailscaled; it doesn't appear to be running (sudo systemctl start tailscaled ?)
root@GDR-SVR:~# /etc/init.d/tailscale start
root@GDR-SVR:~# tailscale up --advertise-routes=10.0.0.0/24,192.168.1.0/24,192.168.2.0/24
failed to connect to local tailscaled; it doesn't appear to be running (sudo systemctl start tailscaled ?)
root@GDR-SVR:~#
```

I dont know why, but after extract and start tailscale always showing ```failed to connect to local tailscaled; it doesn't appear to be running (sudo systemctl start tailscaled ?)``` all depedency was allready installed. Thanks.

dianariyanto, 2022-11-06:
When i try to run ```tailscaled```
```
root@GDR-SVR:~# tailscaled
logtail started
Program starting: v1.32.2-t2f1a945ad-g07d934fd1, Go 1.19.2-ts3fd24dee31: []string{"/tmp/tailscaled"}
LogID: 15948aadd6aea50b8d8c83a0ca14bdc3c22505bab899b2b691ae1e9a5fb726ff
logpolicy: using system state directory "/var/lib/tailscale"
wgengine.NewUserspaceEngine(tun "tailscale0") ...
wgengine.NewUserspaceEngine(tun "tailscale0") error: creating router: exec: "iptables": executable file not found in $PATH
flushing log.
logger closing down
logtail: upload: log upload of 542 bytes compressed failed: Post "https://log.tailscale.io/c/tailnode.log.tailscale.io/f05cf57f31d9d9b99194915190e41ce53864debbbf16f01ff4b408c76bcded75": context canceled
createEngine: creating router: exec: "iptables": executable file not found in $PATH
root@GDR-SVR:~#
```
im using OpenWrt 22.03.2 btw.

adyanth, 2022-11-06:
Doesn't look like you have iptables installed? Try running `opkg install iptables-nft`

dianariyanto, 2022-11-06:
emm.. make sense, the default openwrt 22 now using nft.

closed 2022-11-21 by dianariyanto