adyanth/openwrt-tailscale-enabler #24: tailscale cert error

dieterplex, 2022-11-22:
When generating the certificate, http 500 responded

```console
# tailscale cert openwrt.mydomain.ts.net
500 Internal Server Error: failed to get cert dir
```

system log shows "certDir: no TailscaleVarRoot"
The workaround is adding statedir option to `/usr/bin/tailscaled`

```shell
/tmp/tailscaled --statedir=/tmp "$@"
```

adyanth, 2022-11-22:
Hey @dieterplex Currently the state file is explicitly being set to /etc/config so that the state can be backed up with OpenWRT. Can you confirm if `--statedir` works with `--state` as well?

https://github.com/adyanth/openwrt-tailscale-enabler/blob/98e126ecb731737a2cc072f74c1be71981683188/etc/init.d/tailscale#LL21

adyanth, 2022-11-22:
Also, using /tmp as your stateDir does not persist on reboots. What gets generated there (like certs) and would you need them to persist?

dieterplex, 2022-11-24:
`--statedir` is passed into as `VarRoot`, from [the comment of `VarRoot`](https://github.com/tailscale/tailscale/blob/main/ipn/ipnserver/server.go#L54-L60):
> VarRoot is the Tailscale daemon's private writable
> directory (usually "/var/lib/tailscale" on Linux) that
> contains the "tailscaled.state" file, the "certs" directory
> for TLS certs, and the "files" directory for incoming
> Taildrop files before they're moved to a user directory.
> If empty, Taildrop and TLS certs don't function.

And according to [this behavior](https://github.com/tailscale/tailscale/blob/main/cmd/tailscaled/tailscaled.go#L279-L287)
> If an absolute --state is provided but not --statedir, try to derive a state directory.

Maybe just change the state file to `/etc/tailscale/tailscaled.state` in `/etc/init.d/tailscale` if we want to keep Taildrop files and cert files?
Or append statedir param for a separate `/var/lib/tailscale` dir? Since another copy of crt & key files are save to current working directory when exec `tailscale cert <my-tsnet-domain>`.

When using `/etc/tailscale/tailscaled.state`, listing files would like this:

```
root@OpenWrt:/etc/tailscale# ls -1 *
derpmap.cached.json
tailscaled.log.conf
tailscaled.log1.txt
tailscaled.log2.txt
tailscaled.state

certs:
acme-account.key.pem
openwrt.mydomain.ts.net.crt
openwrt.mydomain.ts.net.key

files:
dieterplex-gmail.com-uid-00000000000000000/
```

Appending both params also works. But to persist on reboots, using `/etc/tailscale` do the trick I think. wdyt?

adyanth, 2022-11-25:
Ah makes sense. I think we can keep the state file as it is (in /etc/config so that it is backed up by openwrt) and set the stateDir to /etc/tailscale so that it persists.

Would you like to send a quick PR?

PR #25 (2022-11-25): Specifying state dir param for generating TLS cert
Fix #24
commit d5c3c789 (2022-11-25): Specifying state dir param for generating TLS cert

Fix #24
closed 2022-11-25 by adyanth