adyanth/openwrt-tailscale-enabler #27: Disable auto-update

zawee-admin, 2023-04-28:
Hi there @adyanth,

Thanks for your work, its all working well on my Teltonika RUTX OpenWRT routers.

How can I disable the auto-update? I'd prefer to update tailscale manually as these devices are in the field.

Thanks

James

adyanth, 2023-04-28:
Removing these lines below will get you that behavior.
https://github.com/adyanth/openwrt-tailscale-enabler/blob/f433061545db16cf7a9baf73f064f19b17b63c27/usr/bin/tailscale#L19-L22

Then, this line will set the version:
https://github.com/adyanth/openwrt-tailscale-enabler/blob/f433061545db16cf7a9baf73f064f19b17b63c27/usr/bin/tailscale#L17

zawee-admin, 2023-04-28:
awesome thanks

closed 2023-04-28 by zawee-admin

adyanth, 2023-04-28:
Remember to do the same on the tailscaled file as well!