adyanth/openwrt-tailscale-enabler #28: Headscale not able to access automatically

peakerz, 2023-05-16:
I applied the following to /etc/rc.local, it is working nice with tailscale .
sleep 15
tailscale up --advertise-exit-node --advertise-routes=192.168.0.0/24,192.168.8.0/24

Now I applied the following to my VPS, it is not working
tailscale up --advertise-exit-node --advertise-routes=192.168.0.0/24,192.168.8.0/24 --login-server=http://<myip>:8080

it seems that tailscale visit tailscle.com regularly, where can I find the setting to switch to my VPS IP?

peakerz, 2023-05-16:
"Does it work on a regular machine with regular tail scale?"  Yes
Is  it possible to DIY tail scale binary ?

adyanth, 2023-05-16:
The binary is not built, it is downloaded from https://pkgs.tailscale.com/stable/
You can try running the command with `/tmp/tailscale` instead of just `tailscale` which bypasses the wrapper.