adyanth/openwrt-tailscale-enabler #29: Unable to establish SSL connection when "tailscale up"

mintisan, 2023-06-17:
```
root@iStoreOS:~# tailscale up --accept-dns=false --advertise-routes=192.168.31.0/24
--2023-06-17 16:39:59--  https://pkgs.tailscale.com/stable/
Resolving pkgs.tailscale.com... 199.38.181.239
Connecting to pkgs.tailscale.com|199.38.181.239|:443... connected.
Unable to establish SSL connection.
Downloading Tailscale _arm64 ..
--2023-06-17 16:39:59--  https://pkgs.tailscale.com/stable/tailscale__arm64.tgz
Resolving pkgs.tailscale.com... 199.38.181.239
Connecting to pkgs.tailscale.com|199.38.181.239|:443... connected.
Unable to establish SSL connection.
gzip: invalid magic
tar: Child returned status 1
tar: Error is not recoverable: exiting now
root@iStoreOS:~# 
```

adyanth, 2023-06-17:
Could you install/update the libustream-openssl and ca-bundle packages and retry? That's the download of Tailscale failing.

mintisan, 2023-06-18:
Yes, It's OK for now today, the file name has version as below. Thanks.

```
root@iStoreOS:~# tailscale up --accept-dns=false --advertise-routes=192.168.31.0/24
--2023-06-18 18:20:59--  https://pkgs.tailscale.com/stable/
Resolving pkgs.tailscale.com... 199.38.181.239
Connecting to pkgs.tailscale.com|199.38.181.239|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'STDOUT'

-                                         [  <=>                                                                     ]  30.51K   128KB/s    in 0.2s    

2023-06-18 18:21:01 (128 KB/s) - written to stdout [31240]

Downloading Tailscale 1.42.0_arm64 ..
--2023-06-18 18:21:01--  https://pkgs.tailscale.com/stable/tailscale_1.42.0_arm64.tgz
Resolving pkgs.tailscale.com... 199.38.181.239
Connecting to pkgs.tailscale.com|199.38.181.239|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 22142063 (21M) [application/x-compressed-tar]
Saving to: 'STDOUT'

-                                       0%[                                                                          ]  14.01K  56.3KB/s               tailscale_1.42.0_arm64/tailscaled
-                                      58%[==========================================>                               ]  12.43M  3.85MB/s    eta 3s     tailscale_1.42.0_arm64/tailscale
-                                     100%[=========================================================================>]  21.12M  5.95MB/s    in 4.0s    

2023-06-18 18:21:06 (5.27 MB/s) - written to stdout [22142063/22142063]

Done!
```

closed 2023-06-18 by mintisan