adyanth/openwrt-tailscale-enabler #31: can not up,when wget

carlsue, 2023-06-22:
[root@xxx:/tmp]#opkg install libustream-openssl ca-bundle kmod-tun
Package libustream-openssl (2016-07-02-ec80adaa-2) installed in root is up to date.
Package ca-bundle (20160104) installed in root is up to date.
Package kmod-tun (3.14.79-1) installed in root is up to date.
[root@xxx:/tmp]#/etc/init.d/tailscale start
[root@xxx:/tmp]#tailscale up --accept-dns=false --advertise-routes=x.x.x.0/
24
Connecting to pkgs.tailscale.com (199.38.181.239:443)
wget: server returned error: HTTP/1.0 400 Bad Request
Downloading Tailscale _mipsle ..
Connecting to pkgs.tailscale.com (199.38.181.239:443)
wget: server returned error: HTTP/1.0 400 Bad Request
tar: invalid magic
tar: short read
[root@xxx:/tmp]#busybox
BusyBox v1.24.1 (2019-02-01 20:53:40 UTC) multi-call binary.
[root@xxx:/tmp]#cat /etc/openwrt_release
DISTRIB_ID="PandoraBox"
DISTRIB_CODENAME="19.02"
DISTRIB_RELEASE="19.02"
DISTRIB_VERSION="4802"
DISTRIB_REVISION="2019-02-01-git-93f2639a7"
DISTRIB_TARGET="ralink/mt7621"
DISTRIB_DESCRIPTION="PandoraBox 19.02 2019-02-01-git-93f2639a7"
DISTRIB_TAINTS="no-all busybox"
DISTRIB_MANUFACTURER="PandoraBox-Team"
DISTRIB_MANUFACTURER="http://www.pandorabox.com.cn"
[root@xxx:/tmp]#cat /etc/openwrt_version
4802

carlsue, 2023-06-22:
i try this ,can work-----vi /usr/bin/tailscale* like this:
   tailscale_version="1.42.0"
    latest_version="$tailscale_version"
[root@xxx:/root]#tailscale up
Downloading Tailscale 1.42.0_mipsle ..
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 20.8M  100 20.8M    0     0  1799k      0  0:00:11  0:00:11 --:--:-- 2377k

carlsue, 2023-06-22:
final:
`#!/bin/sh

set -e

if [ ! -f /tmp/tailscale ]; then
    arch=`uname -m`
    if [ "$arch" == "mips" ]; then
        endianness=`echo -n I | hexdump -o | awk '{ print (substr($2,6,1)=="1") ? "le" : ""; exit }'`
    elif [ "$arch" == "armv7l" ]; then
        arch=arm
    elif [ "$arch" == "aarch64" ]; then
        arch=arm64
    elif [ "$arch" == "x86_64" ]; then
        arch=amd64
    fi

    tailscale_version="1.42.0"

    latest_version="$tailscale_version"

    version="${tailscale_version}_${arch}${endianness}"

    echo "Downloading Tailscale ${version} .."

    echo -e "tailscale_${version}/tailscale" > /tmp/tailscale_${version}_files.txt

    if [ ! -f /tmp/tailscaled ]; then
        echo -e "tailscale_${version}/tailscaled" >> /tmp/tailscale_${version}_files.txt
    fi

    #curl -O -k https://pkgs.tailscale.com/stable/tailscale_${version}.tgz | tar x -zvf - -C /tmp -T /tmp/tailscale_${version}_files.txt
    tar x -zvf /mnt/xxxxx/tailscale_1.42.0_mipsle.tgz -C /tmp -T /tmp/tailscale_${version}_files.txt
    mv /tmp/tailscale_$version/* /tmp
    rm -rf /tmp/tailscale_${version}*

    echo "Done!"
fi

/tmp/tailscale "$@"
`

adyanth, 2023-06-22:
Could you please run this command in the shell and provide the output?

```bash
wget -O- https://pkgs.tailscale.com/stable/ | grep tailscale_ | head -1 | cut -d'_' -f 2
```

Looks like it is getting a 404 trying to pull the version, resulting in an empty version. It works fine for me.

Also, the way you have now is perfectly fine if you are okay with manually updating the version.

carlsue, 2023-06-22:
> Could you please run this command in the shell and provide the output?
> 
> ```shell
> wget -O- https://pkgs.tailscale.com/stable/ | grep tailscale_ | head -1 | cut -d'_' -f 2
> ```
> 
> Looks like it is getting a 404 trying to pull the version, resulting in an empty version. It works fine for me.
> 
> Also, the way you have now is perfectly fine if you are okay with manually updating the version.

[root@xxx:/tmp]#wget -O- https://pkgs.tailscale.com/stable/ | grep tailscale_ |
head -1 | cut -d'_' -f 2
Connecting to pkgs.tailscale.com (199.38.181.239:443)
wget: server returned error: HTTP/1.0 400 Bad Request
it look like the wget have some problem,curl can work standalone,but can not run in your script,when i instad wget with it.

adyanth, 2023-06-22:
Yeah I am not sure why the tailscale website gets a 404. You can comment out the autoupdate lines on both tailscale and tailscaled files and should be good to go.

adyanth, 2023-06-29:
Closing this since the issue is with access to the website.

closed 2023-06-29 by adyanth