adyanth/openwrt-tailscale-enabler #34: Everytime reboot openwrt, tailscale can't work

Yumega, 2023-06-27:
/etc/init.d/tailscale enable  (has enabled)

but Everytime reboot openwrt, the Tailscale admin panel shows openwrt connected (iphone also connected), but my iphone can't connect with openwrt.
but after use this 
 tailscale up 
/etc/init.d/tailscale start
It can connect again.

is there a method to restart tailscale one more time after openwrt rebooted?

Yumega, 2023-06-27:
zerotier doesn't have the problem

adyanth, 2023-06-29:
If the tail scale admin panel shows the device as connected, there is nothing I can suggest to troubleshoot. You should have better luck checking in tail scale or openwrt forums.

Cykaihappy, 2023-07-22:
i met the same trouble .when i restart the tailsacel, tailscale visit https://pkgs.tailscale.com/stable/tailscale_1.46.0_arm64.tgz ,and than reponse 404,
[image]

adyanth, 2023-07-22:
[image]

Looks like 1.46.0 arm64 packages are not yet released.

closed 2023-07-22 by adyanth