adyanth/openwrt-tailscale-enabler #35: Unusual use after openwrt restart

lweixing55, 2023-06-30:
The installation is successful and the use is normal. But after restarting the route, /tmp only downloads tailscaled，but tailscale will not download, and the management background shows that it is connected, other sub-machines cannot pass the exit-node , I don’t know why

adyanth, 2023-06-30:
The first part is expected behavior. Tailscale is not needed once configuration is complete, so only tailscaled is downloaded and started with the configuration stated saved.

Can the router itself access other tailscale devices? I would think either the openwrt is messing with your tailscale routes or tailscale is losing some state. I cant think of a way the latter can happen.

lweixing55, 2023-07-01:
Thank you for your reply! ! The router itself can access other tailscale devices, and other tailscale devices can also access openwrt, that is, the tailscale device cannot pass through the exit-node of the openwrt router

lweixing55, 2023-07-03:
I seem to have found the reason. It was blocked by firewall rules. After deleting the rules and restarting the firewall, the problem was solved. thanks again!

closed 2023-07-03 by lweixing55