adyanth/openwrt-tailscale-enabler #36: bug

lxx12445, 2023-07-03:
There's a mistake in DNS after my openwrt reboot firstly.

adyanth, 2023-07-22:
Could you please elaborate on the issue?

lxx12445, 2023-07-25:
Hello.The following is the problem's detail.My os is Istoreos(based on lede,x86)(https://www.istoreos.com/).In the beginning,I installed tailscale under the guidance of the article on the website(https://blog.51cto.com/fxn2025/5482046).After the installation was complete,it works normally.However,after I rebooted my computer,dns sever was broken.After I pinged github.com and www.baidu.com,it just replyed:'bad address.'That is all.Because my Istoreos is uninstalled,I have no logs.Thank you for your reply.

adyanth, 2024-05-04:
This looks more to be a configuration issue. If the tailscale config mismatches with openwrt, it might not be able to access the network.

closed 2024-05-04 by adyanth