adyanth/openwrt-tailscale-enabler #41: Can't install Tailscale - init.d permissions & bin/tailscaled - read-only

HearthCore, 2023-11-12:
ASUSWRT-Merlin RT-AX92U 3004.388.4_0-gnuton1 Wed Oct 25 19:14:25 UTC 2023
timoyi@router:/tmp/home/root# ls
openwrt-tailscale-enabler-v1.36.1-fb2f6cf-autoupdate.tgz
timoyi@router:/tmp/home/root# tar x -zvC / -f openwrt-tailscale-enabler-v1.36.1-fb2f6cf-autoupdate.tgz
usr/
usr/bin/
usr/bin/tailscaled
tar: can't remove old file usr/bin/tailscaled: Read-only file system
timoyi@router:/tmp/home/root# /etc/init.d tailscale start
-sh: /etc/init.d: Permission denied

---

RT-AX92U (aarch64) FW-388.4 @ 10.0.0.1
 Operation Mode: Wireless router

 amtm - the Asuswrt-Merlin Terminal Menu

 /mnt/router Size 28.3G Used 86.3M (0%)

 ep manage   Entware packages

 fd run      Format disk         fdl show log

 sh manage   shell history
 cj show     all cron jobs
 i  show     all available scripts or tools
 u  check    for script updates

    amtm options
 e  exit      t  theme   r  reset   a  about
_____________________________________________

 Enter amtm option  ep
_____________________________________________

 Entware package options

 This router runs Entware aarch64-k3.10
 Server in use: bin.entware.net

 See available packages list here:
 https://bin.entware.net/aarch64-k3.10/Packages.html

 1. Check for updated Entware packages
 2. Show installed Scripts and Entware packages
 3. Select Entware server to use
 4. Remove Entware

 Enter selection [1-4 e=Exit] 2

 Installing required Entware package 'column'
 for a better file presentation.

Downloading https://bin.entware.net/aarch64-k3.10/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Installing column (2.39-2) to root...
Downloading https://bin.entware.net/aarch64-k3.10/column_2.39-2_aarch64-3.10.ipk
Installing libsmartcols (2.39-2) to root...
Downloading https://bin.entware.net/aarch64-k3.10/libsmartcols_2.39-2_aarch64-3.10.ipk
Configuring libsmartcols.
Configuring column.

_____________________________________________

 List of installed Entware packages (9)

 ca-bundle - 20230311-1  libc - 2.27-11          libpthread - 2.27-11    libsmartcols - 2.39-2   tailscale - 1.46.1-1
 column - 2.39-2         libgcc - 8.4.0-11       librt - 2.27-11         libssp - 8.4.0-11

 Entware Apps installed in /opt/bin/ (9)

 column                  entware-services        grep                    localedef.new           tailscaled
 egrep                   fgrep                   locale.new              tailscale

 Entware Apps installed in /opt/sbin/ (0)

Did i miss a prequesite somewhere?

adyanth, 2023-11-12:
Merlin is not OpenWRT. Looks like it mounts system files in a read only partition.

NikolaRusakov, 2023-12-21:
Hi @HearthCore,

check out my fork, I hope it will give you some ideas how to tackle it.
I run same HW, **RT-AX92U**, same firmware version.

https://github.com/NikolaRusakov/openwrt-tailscale-enabler

firstly, you have to extract you executables in persistent path, where your USB stick / HDD resides.

secondly, (_I haven't managed to make it work on my end yet_);
you'll need to hook your startup service script in path for example `/jffs/addons/tailscale` in [`jffs/scripts/services-start`](https://github.com/RMerl/asuswrt-merlin.ng/wiki/User-scripts) file.

For now, I run tailscaled manually in bg and then `tailscale` with subnet param. Works flawlessly.

Let us know if you make a progress on your side.

Cheers

closed 2024-05-04 by adyanth