adyanth/openwrt-tailscale-enabler #42: how to fix the client version != server version

MuMu360121, 2023-11-14:
run the command then happen like this:
[image]

adyanth, 2023-11-14:
Could you delete the `/tmp/tailscale*` and try again? I am not familiar with iStoreOS.

BJReplay, 2023-11-14:
I usually see that error (client version != server version) immediately after updating versions, when an old version was already loaded in memory and I've downloaded a new version, and I run a tailscale command like tailscale up.

In this case, does iStoreOS include a very early build of Tailscale?

It looks like it does - https://openwrt.org/packages/pkgdata/tailscale that refers to Version:1.32.3-1 and https://openwrt.org/docs/guide-user/services/vpn/tailscale/start make it look like iStoreOS may have included this very early build of this package.

Otherwise, is it possible that you first tried installing this version?

If that is the case, I suggest that before you use this package, you need to 

`opkg remove tailscale`

However, if it is built into the iStoreOS rom, you may be stuck with that version.

MuMu360121, 2023-11-14:
thanks for the answer ，it's worked after I do what you say ,but the exit node can't start
[image]
whatever I'm happy that it can be used, thanks again

BJReplay, 2023-11-14:
You may need to add to your command to get it to run as an exit node:

`tailscale up --accept-dns=false --advertise-routes=10.0.0.0/24 --advertise-exit-node`

You may get a warning

```
Error: changing settings via 'tailscale up' requires mentioning all
non-default flags. To proceed, either re-run your command with --reset or
use the command below to explicitly mention the current value of
all non-default settings:
```

So run again

`tailscale up --accept-dns=false --advertise-routes=10.0.0.0/24 --advertise-exit-node --reset`

Finally, these two articles have some useful instructions around creating interfaces and fw rules.

- https://openwrt.org/docs/guide-user/services/vpn/tailscale/start
- https://www.reddit.com/r/Tailscale/comments/yrqbwb/how_to_using_openwrt_as_a_tailscale_exit_node/

These two options also require you to add `--netfilter-mode=off` to your tailscale up command so it will look like

`tailscale up --accept-dns=false --advertise-routes=10.0.0.0/24 --netfilter-mode=off --advertise-exit-node --reset`

closed 2024-05-04 by adyanth