adyanth/openwrt-tailscale-enabler #48: How to enable the exit-node function in this installation method? I followed the configuration of other tutorials online and used the tailscale up -advertise-exit-node command to enable it. It also showed that it was enabled, but the client could not use the exit function of the node.

6025278, 2024-05-11:
How to enable the exit-node function in this installation method? I followed the configuration of other tutorials online and used the tailscale up -advertise-exit-node command to enable it. It also showed that it was enabled, but the client could not use the exit function of the node.

adyanth, 2024-05-11:
From the tailscale perspective, that would be it. As long as the admin panel shows it to be enabled, it is. 

When you say the client could not use the function, is it unable to select the node as an exit node, or is it selected but no traffic flows through? If it is the former, that would be an issue with the tailscale client/admin panel config, if it is the later, that would be firewall configs on OpenWRT.

6025278, 2024-05-11:
> From the tailscale perspective, that would be it. As long as the admin panel shows it to be enabled, it is.
>
> When you say the client could not use the function, is it unable to select the node as an [...]

The official control panel shows that the exit node function is on, and the Android client can also start and select the exit node normally, but there is no traffic. There is no manual configuration of the firewall in your configuration file, and I did not configure the firewall, which is also the default. I also configured the tailscale firewall according to the online teaching video, but the result is still invalid. Can you teach me how to configure the firewall under these circumstances? Thank you

6025278, 2024-05-11:
After your configuration, it is possible to access openwrt and its internal network devices from the external network by default, which is the exit-node function of openwrt, which does not work on the client.