adyanth/openwrt-tailscale-enabler #51: Tailscale not getting downloaded

sunyoung1, 2024-07-13:
Hello, i'm running to an error which tailscale fails to complete the download for the setup
i have attached the logs below

`root@ax3000t:~# tailscale up --accept-dns=false --advertise-routes=10.
0.0.0/24
Downloading 'https://pkgs.tailscale.com/stable/'
Connecting to 2607:f740:f::d19:443
(null)                   0   - stalled -
Connection reset prematurely
Downloading Tailscale _arm64 ..
Downloading 'https://pkgs.tailscale.com/stable/tailscale__arm64.tgz'
Connecting to 2607:f740:f::d19:443
(null)                   0   - stalled -
Connection reset prematurely
tar: invalid magic
tar: short read
`

moyunkz, 2024-08-20:
I got the same error as you. Here's how I solved it:

1. Edit the file `vi /usr/bin/tailscale `
2. Replace line 34:

`    wget -O- https://pkgs.tailscale.com/stable/tailscale_${version}.tgz | tar x -zvf - -C /tmp -T /tmp/tailscale_${version}_files.txt
`
with
`    curl -LJO https://pkgs.tailscale.com/stable/tailscale_${version}.tgz
    tar -xzvf tailscale_${version}.tgz -C /tmp -T /tmp/tailscale_${version}_files.txt`

Hope this helps!

flier268, 2024-10-13:
```bash
 wget -O- https://dl.tailscale.com/stable/tailscale_${version}.tgz | tar x -zvf - -C /tmp -T /tmp/tailscale_${version}_files.txt
```