adyanth/openwrt-tailscale-enabler #53: Filed enable tailscale at boot

gemnioo, 2024-09-21:
when power-off and reboot. Tailscale failed to connected.  After manual 'tailscale up'. it will auto downloading latest version even manual 'tailscale set --auto-update=false' 

OpenWrt R24.9.9   Kernel 5.10.225 

https://github.com/kissyouhunter/Openwrt_X86-Openwrt_N1-Armbian_N1/releases/tag/n1_openwrt

gemnioo, 2024-09-21:
it need manual tailscale up but shows error

```
failed to connect to local tailscaled; it doesn't appear to be running
```

Maliwag, 2025-02-20:
I am having the same error in your message above. I cannot get the service to start. I went through all the directories to give full permissions to all and still no luck.

gemnioo, 2025-02-20:
> I am having the same error in your message above. I cannot get the service to start. I went through all the directories to give full permissions to all and still no luck.

just disable tailscale auto update and in manually restart it

```
http://192.168.1.1/cgi-bin/luci/admin/system/startup   \\ luci

ls /etc/init.d/  \\ ssh 
```

closed 2025-02-22 by gemnioo