adyanth/openwrt-tailscale-enabler #55: Warning: client version "1.32.3-1 (OpenWrt)" != tailscaled server version "1.74.1-tccd6bf2f4-g8fce4ce11"

HaoBanLu, 2024-10-06:
**When I execute the command:**
tailscale up --accept-dns=false --advertise-routes=192.168.10.0/24

**Prompt warning**
Warning: client version "1.32.3-1 (OpenWrt)" != tailscaled server version "1.74.1-tccd6bf2f4-g8fce4ce11"

**Current environment:**
istoreos

BJReplay, 2024-10-06:
You have installed the opkg tailscale at some point, and also attempted to install this version.  You need to remove the opkg version before attempting to install this version.

HaoBanLu, 2024-10-07:
> You have installed the opkg tailscale at some point, and also attempted to install this version. You need to remove the opkg version before attempting to install this version.

Thank you, you are right, follow the steps you said, I succeed

closed 2024-10-07 by HaoBanLu